ros-industrial/staubli_val3_driver #1: Velocity Override does not have correct effect

marshallpowell97, 2020-01-16:
https://github.com/ros-industrial/staubli_experimental/blob/b4a0d23e54c7f48b5f043314517d782e1cc344de/staubli_val3_driver/val3/ros_server/pushMotion.pgx#L72-L79
Sets the velocity to the value set on the teach pendant if 'velocity override' is enabled. This does not seem to work as intended since the velocity set on the pendant already "overrides" whatever value is set in the Val3 code. The result is that the "overridden" velocity ends up being much smaller than intended.

I.E. if 50% velocity is set by the pendant, the actual speed of the robot will be 50% (result from `getMonitorSpeed()` ) multiplied by another 50% (set by the pendant) = 25%

gavanderhoorn, 2020-03-12:
As I wrote in https://github.com/ros-industrial/staubli_val3_driver/pull/8#discussion_r369798953, I believe it'd be best to remove the override: it reportedly doesn't work as intended.

I'll change the title of this issue to convert it into an action item.

marshallpowell97, 2020-03-12:
I have this removed in a branch [here](https://github.com/marshallpowell97/staubli_val3_driver/tree/removeVelOverwrite) ready to make a pull request whenever the previous one is approved.

gavanderhoorn, 2020-03-12:
It would probably be good to rebase your new branch once #8 gets merged.

gavanderhoorn, 2020-03-30:
I've linked this to #9, as it (starts to) remove(s) the velocity override.

gavanderhoorn, 2020-03-30:
Seeing issues like #2 (and the attempts to fix this in #8 and then #18), perhaps the rationale behind the velocity override functionality was exactly the problems we're encountering now?

closed 2020-06-09 by gavanderhoorn