ros-industrial/staubli_val3_driver #2: High jerk at end of trajectories

gavanderhoorn, 2020-01-16:
Reported on ROS Answers: [Industrial Robot Client sends wrong velocity on last waypoint?](https://answers.ros.org/question/338545).

> Created a plan to rotate single joint from 0 to pi/2 radians. Examined plan looks fine with smoothly tapering velocity, however, actual robot motion ends with accelerated jerk at end. Wireshark packet sniffing reveals velocity in broadcast packet for the last waypoint is wrong.

It's unclear at the moment whether this is an issue in `staubli_val3_driver`, `industrial_robot_client` or somewhere else.

gavanderhoorn, 2020-02-09:
@marshallpowell97: could you perhaps provide some advice on whether by default the *Nominal* or *Maximum* joint velocity limits should be used?

Would be good to have some conclusion here.

marshallpowell97, 2020-02-10:
@gavanderhoorn From what I've been able to gather, the maximum values are a "best case" scenario, and cannot be expected from multiple joints at the same time. The motion descriptor only has a singular "vel" parameter which is applied to the joint that is moving the furthest and all of the other joints will move at a percentage of that. So using the nominal velocities would be my recommendation.

marshallpowell97, 2020-02-10:
Also, side note, I was able to replicate the trajectory above at one point, but now something has changed and my new trajectory looks super messy:
[image]
Any ideas what I could have done? I am using the nominal velocities and accel limits of 9999. (This is not using any of the changes from #8 )

gavanderhoorn, 2020-02-10:
Unfortunately, without looking into this I wouldn't know.

I would recommend to take a look at the ROS side of things though (ie: `plot juggler` and `rostopic echo` of the goal topic).

Looks like lots of acceleration/deceleration?

gavanderhoorn, 2020-03-12:
@marshallpowell97: have you had any opportunity to look into the trajectory execution performance? The plots you show indeed don't look very nice.

gavanderhoorn, 2020-03-30:
Even though #8 was merged, I've not closed this as I believe we've not actually resolved it.

The velocity interpolator added in #8 seems to not work correctly (tracking in #18), so I expect the same problem to exist.

Summarising: the last segment appears to be given a higher value for the `velocity` than the immediately preceding point in some cases, which leads to a sudden jump in TCP velocity and acceleration.

See https://github.com/ros-industrial/industrial_core/issues/247 for my reasoning.

@marshallpowell97: would it perhaps be an idea to use `duration`, instead of `velocity`? Just to see whether this improves the behaviour of the robot?

Oguked, 2022-04-06:
We have the same issue with our TX2-60L... still no solution?

gavanderhoorn, 2022-04-28:
According to the analysis performed in the earlier comments, the problem is not caused by anything here in `staubli_val3_driver`, but in the trajectory relay in `industrial_robot_client`.

As no one has spend time fixing the way in which the trajectory relay converts `JointTrajectory` timing-information into the Simple Message representation, there has been no change to this behaviour.

Another factor could be what was reported in https://github.com/ros-industrial/staubli_val3_driver/issues/2#issuecomment-563364169.

leo002, 2024-08-16:
> Summarising: the last segment appears to be given a higher value for the `velocity` than the immediately preceding point in some cases, which leads to a sudden jump in TCP velocity and acceleration.

For what ever reason, the moveit planner seems to send a default speed for the last segment of the motion, to the joint_trajectory_action/goal topic, which directly controls the robot.

By downloading the industrial_core package and fixing the following line solved the problem for me:

file: industrial_core/industrial_robot_client/include/industrial_robot_client/joint_trajectory_interface.h
Line 72: JointTrajectoryInterface() : default_joint_pos_(0.0), default_vel_ratio(0.1), default_duration_(10.0){}
change default_vel_ratio_(0.1) to default_vel_ratio_(0.01)