ros-industrial/staubli_val3_driver #10: VAL 3 linter?

gavanderhoorn, 2020-01-27:
Just wondering: @marshallpowell97: would you happen to know whether there is a VAL 3 linter available somewhere?

This might be nice to have in the CI setup (#9). But would have to be stand-alone of course (ie: not integrated into SRS).

gavanderhoorn, 2020-01-27:
Seeing as it's all XML (at a low-level), we could run a generic XML linter/parser just to see whether files don't contain any illegal entities, etc.

marshallpowell97, 2020-02-11:
> Just wondering: @marshallpowell97: would you happen to know whether there is a VAL 3 linter available somewhere?

I don't believe so. There is a syntax checker built into SRS, but I am not aware of anything standalone.

gavanderhoorn, 2020-02-12:
> There is a syntax checker built into SRS

is the VAL 3 grammar available somewhere? Either in documentation or as a file?

What does the SRS use to check the syntax? Perhaps some of that could be reused.

marshallpowell97, 2020-02-12:
I did some more digging and I found an "offline syntax checker" that is included with SRS as a `.exe` file. The documentation is lacking, but does this seem to be what you are looking for?
[image]

gavanderhoorn, 2020-02-12:
Oh, that's a nice find.

Would you happen to know what sort of dependencies it has?

We could potentially integrate it into a Windows based CI setup (using github actions fi).

gavanderhoorn, 2020-03-12:
@marshallpowell97: have you been able to find out anything more?

marshallpowell97, 2020-03-19:
@gavanderhoorn I can't seem to get it to work. I am trying to find someone at Staubli that knows how to use it.

If you have access to SRS (even without a license), it should already be included.

gavanderhoorn, 2020-03-30:
>  I am trying to find someone at Staubli that knows how to use it.

have you been able to find anyone?

marshallpowell97, 2020-05-06:
@gavanderhoorn I was able to get this working. Do you have access to a copy of SRS? Even without a license, you should be able to download and install emulators which include the syntax checker for that specific controller/VAL 3 version.

gavanderhoorn, 2020-05-08:
I have some old copies here. I'd have to check whether I could download newer ones.

If not, I'll try to reach out to Staubli Benelux.