ros-industrial/staubli_val3_driver #14: How does one transfers contents of val3 to controller via USB ?

jediofgever, 2020-03-05:
Hi,
I am trying to copy the driver to controller. But cannot find detailed instructions to do so. I can see the folders of `val3` when I plug in USB to controller, but I couldn't figure how to copy those two folders.
this https://github.com/ros-industrial/staubli_val3_driver/issues/6  issue seems to be the address for my problem  but long time no activity there too.  As I am just starting with staubli controller, I can take part in https://github.com/ros-industrial/staubli_val3_driver/issues/6 if there is need

gavanderhoorn, 2020-03-05:
Do you have access to the relevant Staubli manuals?

jediofgever, 2020-03-05:
I do but there is no relevant sections, I am using a CS9 controller btw

closed 2020-03-06 by jediofgever

gavanderhoorn, 2020-03-07:
Did you figure it out?

jediofgever, 2020-03-08:
I did transfer driver to controller but not with USB, unfortunately I had to install Their SRS software on a windows machine,  I think it would be much better If we include the transferring via USB in details. Since most of us ROS users do not operate a windows machine. I still haven’t found related piece of info for  transferring data via USB on staubli manuals