ros-industrial/staubli_val3_driver #20: On-line joint motion control using VAL 3 driver

sonny-tarbouriech, 2020-04-30:
Hi there,

I'm trying to find a solution to perform a reactive control on a Staubli tx2 robot using ROS. From what I understand, the driver provides a FollowJointTrajectory action server which intends to execute a pre-calculated joint trajectory on the robot. 

Would there be a way to use/implement a "low" level controller allowing to execute dynamic motions instead of pre-defined trajectories? 

Thank you in advance for your help.

Sonny

gavanderhoorn, 2020-05-08:
> From what I understand, the driver provides a FollowJointTrajectory action server which intends to execute a pre-calculated joint trajectory on the robot.

Yes, that is correct.

> Would there be a way to use/implement a "low" level controller allowing to execute dynamic motions instead of pre-defined trajectories?

Well, not without changes to the source.

The modifications needed would be similar to ros-industrial/motoman#215, which essentially adds a topic interface which accepts individual points.

No one has released this for the generic clients though, which this driver uses.

The controller itself would certainly support it -- as long as the motion queue is kept non-empty, motions should remain continuous. What you could expect in terms of latency I couldn't predict.

sinamr, 2020-08-06:
Hi, 

In case one do the modifications,  I was wondering what would be the maximum update rate for on-line planning? Is it 250 Hz?

Best,
Sina

gavanderhoorn, 2020-08-06:
I can't say anything conclusively.

Perhaps @marshallpowell97 could comment on that.

gavanderhoorn, 2020-08-06:
Also: closing this due to inactivity.

closed 2020-08-06 by gavanderhoorn

sinamr, 2020-08-06:
Thanks a lot. 
@marshallpowell97 , I just need an approximate number, no need for the accurate one.

marshallpowell97, 2020-08-19:
@sinamr The arm uses a boxcar filter which changes the update frequency of the arm. The filter length is not necessarily the same on each arm, so the update frequency will not always be the same. There are ways to bypass the boxcar filter (I.E. using a pre-calculated spline trajectory) but all require an additional license from Staubli.

I looked into adding something like this to the driver, but it did not make sense to add something here that would only be usable with an additional (paid) license.