ros-industrial/staubli_val3_driver #22: VAL3 version compatibility wtih versions lower than 7.3

rishabhgks, 2020-08-18:
I will be purchasing a used Staubli TX-90L and the seller has informed me that the VAL3 version supported by that controller goes up to version 7.3
This is my first time working on a Staubli arm and on VAL3.
I wanted to know if there is any kind of compatibility between the val3_driver and older versions of VAL3 as the Readme mentioned that the VAL3 version needs to be 7.7.2 atleast.
If it is not compatible, could you guide me on how I should go ahead to solve this issue?

kalihamilton, 2020-08-18:
Wanted to add to this and ask if anyone knows how to update the version of VAL3? Or is that something that has to go through Staubli?

gavanderhoorn, 2020-08-19:
> I wanted to know if there is any kind of compatibility between the val3_driver and older versions of VAL3 as the Readme mentioned that the VAL3 version needs to be 7.7.2 atleast.

We've only tested it with the version of VAL3 mentioned in the readme.

That version was needed as it comes with some of the required socket communication primitives the driver needs.

I cannot say anything about compatibility with older versions of VAL3.

> If it is not compatible, could you guide me on how I should go ahead to solve this issue?

I would suggest to contact Staubli about this, or post your question on a forum dedicated to talking about Staubli robots and controller hardware.

We try to reserve this issue tracker for issues with the `staubli_val3_driver` package hosted here.

gavanderhoorn, 2020-08-19:
> Wanted to add to this and ask if anyone knows how to update the version of VAL3? Or is that something that has to go through Staubli?

I would indeed suggest to contact your local Staubli branch office.

gavanderhoorn, 2020-08-19:
As this is not something actionable here, I'm going to close this issue.

Feel free to keep commenting on it of course.

closed 2020-08-19 by gavanderhoorn

marshallpowell97, 2020-08-19:
>I wanted to know if there is any kind of compatibility between the val3_driver and older versions of VAL3 as the Readme mentioned that the VAL3 version needs to be 7.7.2 atleast.

The readme gives the reason that "this implementation uses return values of sioGet() only available from s7.7.2 onwards". I am not sure that this is accurate since the release notes for VAL3 s7.7.2 do not mention anything about `sioGet()` and such a major change would not usually occur in a minor revision number. The driver passes a syntax check for s7.3 but I cannot confirm that it works at the moment (I will try to get around to it).

@rishabhgks what type of controller does the arm use? There isn't really a way to update the VAL3 version without contacting your local Staubli branch.

kalihamilton, 2020-08-19:
@marshallpowell97 I also didn't see any changes in functionality between sioGet() in Version 6 vs Version 7. 

I did notice the addition of brackets in the sioGet() definition in the documentation (reference manuals from tech support database): 
v6: num sioGet(sio siInput, num& nData)
v7, v8: num sioGet(sio siInput, num& nData[])

as well as the call to the serial connection is also different: 
v6: nb = sioGet(io:TcpServer, nByteIn[i])
v7: nb = sioGet(siTcpServer, nByteIn[i])

Not sure if these are significant enough changes to make the older versions incompatible though...

gavanderhoorn, 2020-08-19:
It's been a while (the original code was contributed in 2015), but I believe the issue was that the behaviour of `sioGet(..)` an `sioSet(..)` was different between the two versions we tested (one 7.7.2 and the other an older one). I've tried to find my emails about this, but I can't find a single one which really explains what we did or why any more.

If it turns out older versions of VAL3 also work, great. We'll just update the readme.