ros-industrial/staubli_val3_driver #24: Unable to connect to live robot

theotromp97, 2020-11-23:
This is my first post, so i am sure i am missing some information.

I am trying to connect to a real staubli rx160 robot, but cannot get the communication working.

**What i have done so far:**
I have build the staubli rx160 package
I have build the staubli_val3_driver package and followed instructions

I can launch the moveit config with a fake staubli robot and move to random coordinate (plan and excecute).

I have configured the TCP ports as follows:
[image]

I can run the program on the staubli robot and it says the following (image is emulator, but i get the same image on the real robot):
[image]

I have configured the ip_address of the robot and laptop in the same range. I can ping the robot.
Laptop ip: 192.168.1.200
Robot ip: 192.168.1.5

When i launch the following:
`roslaunch staubli_val3_driver robot_interface_streaming.launch robot_ip:='192.168.1.5'`

i get the following error:

> ... logging to /home/x/.ros/log/e6740e48-2d98-11eb-8bfe-000c291fbc40/roslaunch-ubuntu-13161.log
> Checking log directory for disk usage. This may take a while.
> Press Ctrl-C to interrupt
> Done checking log file disk usage. Usage is <1GB.
> 
> started roslaunch server http://ubuntu:45455/
> 
> SUMMARY
> ========
> 
> PARAMETERS
>  * /robot_ip_address: 192.168.1.5
>  * /rosdistro: melodic
>  * /rosversion: 1.14.10
> 
> NODES
>   /
>     joint_trajectory_action (industrial_robot_client/joint_trajectory_action)
>     motion_streaming_interface (industrial_robot_client/motion_streaming_interface)
>     robot_state (industrial_robot_client/robot_state)
> 
> auto-starting new master
> process[master]: started with pid [13171]
> ROS_MASTER_URI=http://localhost:11311
> 
> setting /run_id to e6740e48-2d98-11eb-8bfe-000c291fbc40
> process[rosout-1]: started with pid [13182]
> started core service [/rosout]
> process[robot_state-2]: started with pid [13189]
> process[motion_streaming_interface-3]: started with pid [13190]
> process[joint_trajectory_action-4]: started with pid [13191]
> [ WARN] [1606142029.588070245]: Unable to find user-specified joint names in 'controller_joint_names'
> [ WARN] [1606142029.588861400]: Unable to find URDF joint names in 'robot_description'
> [ERROR] [1606142029.588939423]: Cannot find user-specified joint names. Tried ROS parameter 'controller_joint_names' and the URDF in 'robot_description'.
> [ERROR] [1606142029.588978764]: Failed to initialize joint_names.  Aborting
> [ WARN] [1606142029.591819220]: Unable to find user-specified joint names in 'controller_joint_names'
> [ WARN] [1606142029.592067259]: Unable to find URDF joint names in 'robot_description'
> [ERROR] [1606142029.592114193]: Cannot find user-specified joint names. Tried ROS parameter 'controller_joint_names' and the URDF in 'robot_description'.
> [ERROR] [1606142029.592125622]: Failed to initialize joint_names.  Aborting
> [ WARN] [1606142029.595985637]: Unable to find user-specified joint names in 'controller_joint_names'
> [ WARN] [1606142029.596864161]: Unable to find URDF joint names in 'robot_description'
> [ERROR] [1606142029.596892462]: Cannot find user-specified joint names. Tried ROS parameter 'controller_joint_names' and the URDF in 'robot_description'.
> [ERROR] [1606142029.596921317]: Failed to initialize joint_names.
> [robot_state-2] process has finished cleanly
> log file: /home/install/.ros/log/e6740e48-2d98-11eb-8bfe-000c291fbc40/robot_state-2*.log
> [ WARN] [1606142030.602095761]: Trajectory state not received for 1.000000 seconds

Hope my information is sufficient!
Thanks in advance

Theo

gavanderhoorn, 2020-11-23:
The error message:

```
[ WARN] [1606142029.588070245]: Unable to find user-specified joint names in 'controller_joint_names'
[ WARN] [1606142029.588861400]: Unable to find URDF joint names in 'robot_description'
[ERROR] [1606142029.588939423]: Cannot find user-specified joint names. Tried ROS parameter 'controller_joint_names' and the URDF in 'robot_description'.
```

gives us an indication of what is going "wrong": the driver needs to know how to *map* incoming joint state messages (from the robot), which are essentially just arrays with anonymous `double`s, to ROS joint names, as `JointState` messages work with *names*, not anonymous arrays of `double`s).

Without supplying it with some information, the driver doesn't have that map, and cannot correctly perform the mapping.

That's what it's telling you: it either gets that information by reading the `controller_joint_names` ROS parameter, or by looking at your URDF (in the `robot_description` parameter) and then *assuming* the order in which your joints are specified in the URDF is the order in which the information from the robot comes in (ie: in those arrays of anonymous `double`).

If you start the driver in isolation, you won't have that parameter set, nor have loaded a URDF, hence: error.

You'll need to load a URDF *before* starting the driver, *or* set the `controller_joint_names` parameter.

I'd actually recommend the latter, as it will make your configuration more robust: it won't depend on how the URDF happens to be structured.

We don't have any examples of this here, or in `ros-industrial/staubli` or `ros-industrial/staubli_experimental`, but take a look at [fanuc_m10ia_support/launch/robot_interface_streaming_m10ia.launch](https://github.com/ros-industrial/fanuc/blob/fda46dde4580ef5a94728b9e349a094203dba207/fanuc_m10ia_support/launch/robot_interface_streaming_m10ia.launch). That would be essentially the structure you could use (of course you'd `include` the `staubli_val3_driver robot_interface_streaming.launch` file, not the fanuc one).

gavanderhoorn, 2020-11-23:
As this is not an issue with the driver itself, but a configuration one, I'm going to close this issue.

It could be said we'd need to update / extend the documentation, and that would be true, but we'd need to create a tracking issue for that, not keep this one open.

We'd also need community contributions to create said documentation.

In the meantime: please feel free to keep commenting on your issue @theotromp97.

If it turns out we do need to fix something here, we'll re-open.

closed 2020-11-23 by gavanderhoorn