ros-industrial/staubli_val3_driver #25: trajectory stops without error message

Aurelien-rkz, 2020-12-02:
Hello everyone,

I'm currently working on a project which consists in doing 3d printing with a staubli robot. To do this I have a list of points that I cut into several trajectories. To compute the trajectories I use "cartesian Path" and the OMPL library but I also tried CHOMP and the Descartes project.

The majority of these trajectories work correctly but the problem is that some of my trajectories don't finish, and I don't get any error messages. The robot simply stays in place.

**My configuration :** 
Robot : Staubli Tx2 60L
Controller: Staubli CS9
Ubuntu 18.04.5 LTS
ROS Melodic

**Remarks :**
The points are accessible by the robot 
The simulation under gazebo works correctly.
I also tried to shift all the points on x and y but the problem remains the same.
On the other hand if I make smaller trajectories (less than 10 points) the problem does not occur.

Attached points are an example of a problematic trajectory.
[points.pdf](https://github.com/ros-industrial/staubli_val3_driver/files/5628288/points.pdf)

Thank you for your help and sorry if there is a lack of information, this is my first post.

Aurelien

gavanderhoorn, 2020-12-02:
In order to be able to say anything, we're going to need more information.

Please provide the following:

 1. a full copy-paste of the output on your console after `roslaunch`ing your system (append `--screen` to the `roslaunch` command so all the output is shown)
 1. the ROS `JointTrajectory` which gets submitted as part of the action goal to the driver
 1. a capture of the network traffic between the ROS pc and the CS9

The trajectory you should be able to capture using `rostopic echo` on the topic which carries the goal message.

Network traffic can be captured using Wireshark.

> I'm currently working on a project which consists in doing 3d printing with a staubli robot

the current implementation of the driver is not the best imaginable, so I'd be interested to hear about your experiences wrt resolution and performance.

Aurelien-rkz, 2020-12-02:
I increased the value of the "eef_step" parameter and that seems to have solved the problem. On the other hand the speeds of the trajectories are not constant, I also have the problem described in issue #2 .

> the current implementation of the driver is not the best imaginable, so I'd be interested to hear about your experiences wrt `resolution` and performance.

I'll let you know the results when I've done my first tests.

gavanderhoorn, 2020-12-02:
> I increased the value of the "eef_step" parameter and that seems to have solved the problem.

So you did not check the return value (ie: `fraction`) to see whether `computeCartesianPath(..)` was able to complete interpolation of your paths?

gavanderhoorn, 2020-12-02:
As this appears to not be a problem with the package in this repository, I'm going to close it.

Feel free to keep commenting on it.

Aurelien-rkz, 2020-12-02:
All right, thanks for your help.

The fraction value was 1.0 but during execution it was still a problem.

gavanderhoorn, 2020-12-02:
> The fraction value was 1.0 but during execution it was still a problem.

Then I'm still interested in seeing the information requested in https://github.com/ros-industrial/staubli_val3_driver/issues/25#issuecomment-737170702.

If `computeCartesianPath(..)` was able to iterate through the full trajectory and output a `JointTrajectory`, then the driver should be able to execute it.

Aurelien-rkz, 2020-12-02:
> 1) a full copy-paste of the output on your console after roslaunching your system (append --screen to the roslaunch command so all the output is shown)

```
printer1@ubuntu:~/ws_moveit/src/staubli_experimental/tx2_60l/Programmes/GUI/AA_erreur$ python main.py
... logging to /home/printer1/.ros/log/663a434e-34bb-11eb-8f8f-000c295b7366/roslaunch-ubuntu-29809.log
started roslaunch server http://ubuntu:37303/

SUMMARY
========

PARAMETERS
 * /controller_joint_names: ['joint_1', 'join...
 * /move_group/Staubli_arm/default_planner_config: 
 * /move_group/Staubli_arm/longest_valid_segment_fraction: 0.005
 * /move_group/Staubli_arm/planner_configs: ['SBL', 'EST', 'L...
 * /move_group/Staubli_arm/projection_evaluator: joints(joint_1,jo...
 * /move_group/allow_trajectory_execution: True
 * /move_group/capabilities: 
 * /move_group/controller_list: [{'action_ns': 'j...
 * /move_group/disable_capabilities: 
 * /move_group/jiggle_fraction: 0.05
 * /move_group/max_range: 5.0
 * /move_group/max_safe_path_cost: 1
 * /move_group/moveit_controller_manager: moveit_simple_con...
 * /move_group/moveit_manage_controllers: True
 * /move_group/octomap_resolution: 0.025
 * /move_group/planner_configs/BFMT/balanced: 0
 * /move_group/planner_configs/BFMT/cache_cc: 1
 * /move_group/planner_configs/BFMT/extended_fmt: 1
 * /move_group/planner_configs/BFMT/heuristics: 1
 * /move_group/planner_configs/BFMT/nearest_k: 1
 * /move_group/planner_configs/BFMT/num_samples: 1000
 * /move_group/planner_configs/BFMT/optimality: 1
 * /move_group/planner_configs/BFMT/radius_multiplier: 1.0
 * /move_group/planner_configs/BFMT/type: geometric::BFMT
 * /move_group/planner_configs/BKPIECE/border_fraction: 0.9
 * /move_group/planner_configs/BKPIECE/failed_expansion_score_factor: 0.5
 * /move_group/planner_configs/BKPIECE/min_valid_path_fraction: 0.5
 * /move_group/planner_configs/BKPIECE/range: 0.0
 * /move_group/planner_configs/BKPIECE/type: geometric::BKPIECE
 * /move_group/planner_configs/BiEST/range: 0.0
 * /move_group/planner_configs/BiEST/type: geometric::BiEST
 * /move_group/planner_configs/BiTRRT/cost_threshold: 1e300
 * /move_group/planner_configs/BiTRRT/frountier_node_ratio: 0.1
 * /move_group/planner_configs/BiTRRT/frountier_threshold: 0.0
 * /move_group/planner_configs/BiTRRT/init_temperature: 100
 * /move_group/planner_configs/BiTRRT/range: 0.0
 * /move_group/planner_configs/BiTRRT/temp_change_factor: 0.1
 * /move_group/planner_configs/BiTRRT/type: geometric::BiTRRT
 * /move_group/planner_configs/EST/goal_bias: 0.05
 * /move_group/planner_configs/EST/range: 0.0
 * /move_group/planner_configs/EST/type: geometric::EST
 * /move_group/planner_configs/FMT/cache_cc: 1
 * /move_group/planner_configs/FMT/extended_fmt: 1
 * /move_group/planner_configs/FMT/heuristics: 0
 * /move_group/planner_configs/FMT/nearest_k: 1
 * /move_group/planner_configs/FMT/num_samples: 1000
 * /move_group/planner_configs/FMT/radius_multiplier: 1.1
 * /move_group/planner_configs/FMT/type: geometric::FMT
 * /move_group/planner_configs/KPIECE/border_fraction: 0.9
 * /move_group/planner_configs/KPIECE/failed_expansion_score_factor: 0.5
 * /move_group/planner_configs/KPIECE/goal_bias: 0.05
 * /move_group/planner_configs/KPIECE/min_valid_path_fraction: 0.5
 * /move_group/planner_configs/KPIECE/range: 0.0
 * /move_group/planner_configs/KPIECE/type: geometric::KPIECE
 * /move_group/planner_configs/LBKPIECE/border_fraction: 0.9
 * /move_group/planner_configs/LBKPIECE/enforce_joint_model_state_space: True
 * /move_group/planner_configs/LBKPIECE/longest_valid_segment_fraction: 0.1
 * /move_group/planner_configs/LBKPIECE/min_valid_path_fraction: 0.5
 * /move_group/planner_configs/LBKPIECE/range: 0.0
[... 1634 lines cut ...]
    time_from_start: 
      secs: 15
      nsecs: 169350430
  - 
    positions: [-1.724804678205732, 0.6794495018782969, 1.5843648458185458, -0.23819424819986418, -0.7071916596811149, 0.18254937174586713]
    velocities: [-0.10883786903095333, -0.09646644216533994, 0.166607415025613, -0.14605343836877155, -0.08772482162776697, 0.10066217015500947]
    accelerations: [0.6063786638010421, 0.5615676242530971, -0.9728681497277359, 0.8280088472586824, 0.5027543497246855, -0.5797580397249976]
    effort: []
    time_from_start: 
      secs: 15
      nsecs: 253784872
  - 
    positions: [-1.7378492464495947, 0.6682226000730367, 1.6037134201057688, -0.25550079531545666, -0.717511262034382, 0.19435116345973832]
    velocities: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
    accelerations: [0.6706841010358762, 0.5772291120652177, -0.9948034238892857, 0.8898129686827497, 0.5305804759467612, -0.6067869720375293]
    effort: []
    time_from_start: 
      secs: 15
      nsecs: 451013905
---
```

> 4) a capture of the network traffic between the ROS pc and the CS9

[image]

gavanderhoorn, 2020-12-03:
> `printer1@ubuntu:~/ws_moveit/src/staubli_experimental/tx2_60l/Programmes/GUI/AA_erreur$ python main.py`

you have a python program starting `roslaunch`?

and I was actually interested in seeing the output of the driver after you submit the trajectory goal.

Did you append `--screen` to the `roslaunch` invocation? If possible, please start things with `roslaunch` directly, instead of using any (Python) wrapper.

> a capture of the network traffic between the ROS pc and the CS9

With this, I meant you'd have to send us the *actual* capture. Not a screenshot.

Capture the traffic with Wireshark, then `File -> Save As` (or `Fichier -> Enregistrez sous`) and attach the resulting file here to a comment.

Without the traffic, I cannot tell you whether the problem is in the driver or not.

Aurelien-rkz, 2020-12-03:
Here is the output of the console from the roslaunch to the moment when the robot remains blocked.

```
printer1@ubuntu:~$ roslaunch tx2_60l streaming.launch robot_ip:=172.17.42.42 --screen
... logging to /home/printer1/.ros/log/8d721152-356b-11eb-a9ce-000c295b7366/roslaunch-ubuntu-3131.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:34141/

SUMMARY
========

PARAMETERS
 * /controller_joint_names: ['joint_1', 'join...
 * /move_group/Staubli_arm/default_planner_config: 
 * /move_group/Staubli_arm/longest_valid_segment_fraction: 0.005
 * /move_group/Staubli_arm/planner_configs: ['SBL', 'EST', 'L...
 * /move_group/Staubli_arm/projection_evaluator: joints(joint_1,jo...
 * /move_group/allow_trajectory_execution: True
 * /move_group/capabilities: 
 * /move_group/controller_list: [{'action_ns': 'j...
 * /move_group/disable_capabilities: 
 * /move_group/jiggle_fraction: 0.05
 * /move_group/max_range: 5.0
 * /move_group/max_safe_path_cost: 1
 * /move_group/moveit_controller_manager: moveit_simple_con...
 * /move_group/moveit_manage_controllers: True
 * /move_group/octomap_resolution: 0.025
 * /move_group/planner_configs/BFMT/balanced: 0
 * /move_group/planner_configs/BFMT/cache_cc: 1
 * /move_group/planner_configs/BFMT/extended_fmt: 1
 * /move_group/planner_configs/BFMT/heuristics: 1
 * /move_group/planner_configs/BFMT/nearest_k: 1
 * /move_group/planner_configs/BFMT/num_samples: 1000
 * /move_group/planner_configs/BFMT/optimality: 1
 * /move_group/planner_configs/BFMT/radius_multiplier: 1.0
 * /move_group/planner_configs/BFMT/type: geometric::BFMT
 * /move_group/planner_configs/BKPIECE/border_fraction: 0.9
 * /move_group/planner_configs/BKPIECE/failed_expansion_score_factor: 0.5
 * /move_group/planner_configs/BKPIECE/min_valid_path_fraction: 0.5
 * /move_group/planner_configs/BKPIECE/range: 0.0
 * /move_group/planner_configs/BKPIECE/type: geometric::BKPIECE
 * /move_group/planner_configs/BiEST/range: 0.0
 * /move_group/planner_configs/BiEST/type: geometric::BiEST
 * /move_group/planner_configs/BiTRRT/cost_threshold: 1e300
 * /move_group/planner_configs/BiTRRT/frountier_node_ratio: 0.1
 * /move_group/planner_configs/BiTRRT/frountier_threshold: 0.0
 * /move_group/planner_configs/BiTRRT/init_temperature: 100
 * /move_group/planner_configs/BiTRRT/range: 0.0
 * /move_group/planner_configs/BiTRRT/temp_change_factor: 0.1
 * /move_group/planner_configs/BiTRRT/type: geometric::BiTRRT
 * /move_group/planner_configs/EST/goal_bias: 0.05
 * /move_group/planner_configs/EST/range: 0.0
 * /move_group/planner_configs/EST/type: geometric::EST
 * /move_group/planner_configs/FMT/cache_cc: 1
 * /move_group/planner_configs/FMT/extended_fmt: 1
 * /move_group/planner_configs/FMT/heuristics: 0
 * /move_group/planner_configs/FMT/nearest_k: 1
 * /move_group/planner_configs/FMT/num_samples: 1000
 * /move_group/planner_configs/FMT/radius_multiplier: 1.1
 * /move_group/planner_configs/FMT/type: geometric::FMT
 * /move_group/planner_configs/KPIECE/border_fraction: 0.9
 * /move_group/planner_configs/KPIECE/failed_expansion_score_factor: 0.5
 * /move_group/planner_configs/KPIECE/goal_bias: 0.05
 * /move_group/planner_configs/KPIECE/min_valid_path_fraction: 0.5
 * /move_group/planner_configs/KPIECE/range: 0.0
 * /move_group/planner_configs/KPIECE/type: geometric::KPIECE
 * /move_group/planner_configs/LBKPIECE/border_fraction: 0.9
[... 663 lines cut ...]
[ INFO] [1607002236.027872401] [/motion_streaming_interface] [ros.industrial_robot_client]: Point[173 of 189] sent to controller
[ INFO] [1607002236.044020269] [/motion_streaming_interface] [ros.industrial_robot_client]: Point[174 of 189] sent to controller
[ INFO] [1607002236.060215584] [/motion_streaming_interface] [ros.industrial_robot_client]: Point[175 of 189] sent to controller
[ INFO] [1607002236.075887750] [/motion_streaming_interface] [ros.industrial_robot_client]: Point[176 of 189] sent to controller
[ INFO] [1607002236.092244836] [/motion_streaming_interface] [ros.industrial_robot_client]: Point[177 of 189] sent to controller
[ INFO] [1607002236.107933154] [/motion_streaming_interface] [ros.industrial_robot_client]: Point[178 of 189] sent to controller
[ INFO] [1607002236.123830011] [/motion_streaming_interface] [ros.industrial_robot_client]: Point[179 of 189] sent to controller
[ INFO] [1607002236.140210851] [/motion_streaming_interface] [ros.industrial_robot_client]: Point[180 of 189] sent to controller
[ INFO] [1607002236.156132382] [/motion_streaming_interface] [ros.industrial_robot_client]: Point[181 of 189] sent to controller
[ INFO] [1607002236.172135712] [/motion_streaming_interface] [ros.industrial_robot_client]: Point[182 of 189] sent to controller
[ INFO] [1607002236.187965458] [/motion_streaming_interface] [ros.industrial_robot_client]: Point[183 of 189] sent to controller
[ INFO] [1607002236.204072494] [/motion_streaming_interface] [ros.industrial_robot_client]: Point[184 of 189] sent to controller
[ INFO] [1607002236.220079660] [/motion_streaming_interface] [ros.industrial_robot_client]: Point[185 of 189] sent to controller
[ INFO] [1607002236.235944224] [/motion_streaming_interface] [ros.industrial_robot_client]: Point[186 of 189] sent to controller
[ INFO] [1607002236.252521186] [/motion_streaming_interface] [ros.industrial_robot_client]: Point[187 of 189] sent to controller
[... cut ...]
[ INFO] [1607002236.283992827] [/motion_streaming_interface] [ros.industrial_robot_client]: Point[189 of 189] sent to controller
[ INFO] [1607002236.289625225] [/motion_streaming_interface] [ros.industrial_robot_client]: Trajectory streaming complete, setting state to IDLE

```
And here you can find the wireshark file.

[Wireshark.lnk.pcapng.zip](https://github.com/ros-industrial/staubli_val3_driver/files/5636403/Wireshark.lnk.pcapng.zip)

Thank you for your help, let me know if it doesn't fit or if anything else can be useful.

gavanderhoorn, 2020-12-03:
Does it start executing the trajectory of 189 points, or does it not even attempt to do that?

The wireshark capture shows the controller returns `FAILURE` for a lot of trajectory points, which is perhaps not being handled correctly the client on the ROS side.

I'll have to check the VAL3 again to see under what conditions the controller will return `FAILURE`.

Could you perhaps also post a photo of the teach pendant when the robot stops moving? The driver should show a status screen on the teach pendant. It would be informative to see a photo of that.

Aurelien-rkz, 2020-12-03:
Yes, he started the 189-point trajectory but got stuck during it.

Here is the picture at the moment of the blockage :

[image]

gavanderhoorn, 2020-12-03:
It's interesting it shows the progress for point `120` to be stuck at `120.9999999999999`. That seems to point to a problem not with the communication between the ROS PC and the controller, but with something getting stuck on the controller (ie: VAL3 side).

@marshallpowell97: would you happen to have any ideas perhaps?

Aurelien-rkz, 2020-12-07:
I noticed that the moveID does not start at 0 but at -1. Is this normal? Because this could explains why point 120 is stuck at 120.9999999999999.

This can be found in the staubli_val3_driver/staubli_val3_driver/val3/ros_server/motionControl.pgx file at the line 27.

[image]

But this does not solve the problem. Here is a picture when the blockage occurs with the moveID starting at 0

[image]

gavanderhoorn, 2020-12-18:
Just to let you know: I've not forgotten about this issue, but I haven't been able to find anything immediately in the code.

As I currently don't have access to a CS9 (or CS8 for that matter), it's difficult for me to test.

I'm still hoping @marshallpowell97 could add something.

> I noticed that the moveID does not start at 0 but at -1. Is this normal?

yes, it's just a trick as trajectories start a `0`, and you're seeing the initial values of the display items here I believe.

> Because this could explains why point 120 is stuck at 120.9999999999999.

I'm not sure I follow. The progress is a direct copy of whatever VAL 3 interpreter returns.

gavanderhoorn, 2021-02-25:
Is this still a problem?