ros-industrial/staubli_val3_driver #26: incomplete while one action client was done

Gloriabhsfer, 2020-12-17:
Hi all:
I am using the staubli TX2-60 (CS9) with ros melodic version. In order to realize a pick and place task, because now the val3 driver will ignore "duration time" while we execute` trajectory_msgs.msg`, and every actionlib client will send one goal to the server( at robot side), I tried to use multi clients under one node, and each client will send robot one goal, in that case, we can realize a break time while robot is running( this break time can make sure our end-effector have time to do the task).
I have meet serval problems while I am doing this:
1. when one client has been done and we want to start the next, the code will be stuck at the client2 and do not execute the waypoints. I have searched online, 2 functions may be useful:1 is cancel_goal( ). I add this under client 1 has been done, but it returned me error said client1 has already been done and no goal need to be cancel. 2 is get_result( ). I print out the result for this function`error_code: 0 error_string: ''` The return seems weird ( maybe need to check val3_driver server part to see the return of this part?)
2. If the driver can fix the problem that it will ignore the duration time parameter and let the robot run waypoints and the start time we set in the duration time parameter, it also works

Here is what I have done in my code [<script src="https://gist.github.com/Gloriabhsfer/0e02f66f5e0881b5c9a2a8dacebaabfd.js"></script>](url)

Any ideas?
Best,
Yu Zhang

gavanderhoorn, 2020-12-18:
I'm not sure I understand what you write.

As a high-level comment:

> because now the val3 driver will ignore "duration time" while we execute` trajectory_msgs.msg`

the driver does not ignore it, but it depends a bit on what you are referring to exactly.

Your link doesn't seem to work.

If you could clarify your post, perhaps we can help.

Gloriabhsfer, 2020-12-18:
Hi Gavanderhoorn:

Thanks for your reply. The gist link can works if you copy it and paste it in a new window. I tried many times but the direct link cannot work and I don't know why.
When we worked on the real robot, we can only control the speed by defining a speed rate at the pendent, the method we use trajectory_msgs will not work ( in the normal situation, we set duration=10, which means it will start at 10s, but staubli robot doesn't behave like this. It will receive a goal from client and calculate the speed defined by the smart pendant and let robot only follow the waypoints you want. ) So I am not very sure if the driver ignores the duration time parameter or I didn't write the right things. But I just followed the syntax that I have done for UR5 and Yaskawa GP8.
`  g.trajectory.points = [
            JointTrajectoryPoint(positions=joints_pos, velocities=[0]*6, time_from_start=rospy.Duration(0.0)),
            JointTrajectoryPoint(positions=q1, velocities=[0]*6, time_from_start=rospy.Duration(2.0)),
            JointTrajectoryPoint(positions=q2, velocities=[0]*6, time_from_start=rospy.Duration(3.0)),
            JointTrajectoryPoint(positions=q3, velocities=[0]*6, time_from_start=rospy.Duration(4.0))]`
For example, code like this part should start the q3 at 4s, but if we set a very slow speed at the pendant, it will take more than 4s to move it. But in the other robot's driver, it will move by calculating the duration time and change speed between every 2 points.
And after the job's in the server has been done, the server will be stuck and could not send another job to its server. When I call "client.get_result" in actionlib, it will return me `"error_code: 0 error_string: '' "` which I am not sure is normal or not, but I assume once job has been done, the result is not something called error_code.

Best,
Gloria