ros-industrial/staubli_val3_driver #27: rosGenericMsg : invalid type

benthebear93, 2021-01-15:
Hello, i'm trying to run ros_sever with SRS.
ros_sever, ros_libs folder are in this directory "C:\Users\MHM\Documents\Staubli\SRS\ros_test2\Controller1\usr\usrapp" 
which is a cell dicrectory.  

I opened ros_sever application and used emulator to run 
+Disk->ros_server 

but error occur and log says "rosGenericMsg: rosGenericMsg : invalid type"

I set the TCP sever two socket (Feedback, Motion) but same error occurs. 
same rosGenericMsg error occurs when i tried real robot. 

Am i missing something? or  is there a problem ?
thx!

SinaMirrazavi, 2021-01-15:
Interestingly, I am also facing the same problem now, 
Any suggestion would be highly appreciated :)

Thanks a  lot

gavanderhoorn, 2021-01-15:
Tbh I don't know immediately what would cause this.

Seeing as you're both running into the same problem, perhaps a good first step would be to provide details on:

 - (virtual) controller versions
 - system software versions
 - VAL3 versions
 - how you retrieved a copy of the VAL3 source
 - how you copied the code to the controller / simulation

etc.

The type is defined here: [ros-industrial/staubli_val3_driver/staubli_val3_driver/val3/ros_server/ros_server.dtx](https://github.com/ros-industrial/staubli_val3_driver/blob/7913f147663f6d54279cfb77cf90ecfb32454313/staubli_val3_driver/val3/ros_server/ros_server.dtx#L16-L33), so it's definitely provided by the sources.

benthebear93, 2021-01-16:
Sorry for late reply

(virtual) controller versions : CS8 (not sure where i can check in virtual env but my real controller version is CS8)
system software versions : SRS 2019.7.2
VAL3 versions : s7.11.2
how you retrieved a copy of the VAL3 source : I download from this page, and put ros_libs, ros_server folder into usr/usrapp folder which exists in cell location. (I'm not sure i understand your question well..)
how you copied the code to the controller / simulation : for controller, i used FTP surfer to put ros_libs, ros_server into usr/usrapp for controller, and simulation is same as above method

I noticed that when i first start the SRS Cell, 7 error occurs.

datatype 'rosTrajPtMsg': type 'JointTrajPtMsg" is unknown.
datatype 'rosTrajPtFMsg': type 'JointTrajPtFMsg" is unknown.
datatype 'rosStatusMsg': type 'RobotStatusMsg" is unknown.
datatype 'rosJointMsg': type 'JointStateMsg" is unknown.
datatype 'rosTrajFbkMsg': type 'JointFbkMsg" is unknown.
datatype 'rosGenericMsg': type 'RosSimpleMsg" is unknown.
datatype 'qMoveBuffer': type 'Queue" is unknown.
(some spelling might wrong since i translate the korean error msg to english)

maybe problem some files from ros_libs ? or i copied wrong place?
(both ros_server folder and ros_libs folder are located C:\Users\MHM\Documents\Staubli\SRS\ros_test2\Controller1\usr\usrapp here)
or do i have to "open application" for all the ros_libs files? 

not sure what is the problem at all, i'd appreciate any tips, ideas. 
If you need more information to figure out i will give it to you. 
thx.

marshallpowell97, 2021-01-18:
Hi @benthebear93 

I can't seem to recreate the error in SRS. Can you run a syntax check in SRS and show me the output?

[image]

benthebear93, 2021-01-19:
@marshallpowell97 
[image]

here is the error

gavanderhoorn, 2021-01-19:
@benthebear93: could you please clarify how you downloaded the files of the driver *exactly*?

If you downloaded any links, please tell us which links.

if you used git, please tell us how.

benthebear93, 2021-01-19:
@gavanderhoorn 

i got from this git. using zip file for download to test. 
also used git clone to test. 

but after you said, i test it again with git clone from readme description and it works.
No error when i used "Check ros_server"
No error using SRS emulator  application manager

I didn't really changed anything. I also reused the old files that i used before today but error didn't show too, weird.
thanks for the answers though, if you need any more information i love to answer.

gavanderhoorn, 2021-01-19:
then the same questions about how you checked out the sources for @SinaMirrazavi.

gavanderhoorn, 2021-01-24:
I'm going to close this.

It seems @benthebear93's problem was resolved.

@SinaMirrazavi: if you're still experiencing problems, please post an update in a comment here.

SinaMirrazavi, 2021-01-24:
Sorry for the late reply. I exactly followed the same procedure benthebear93 described and by cloning the repo the problem was sovled.

gavanderhoorn, 2021-01-24:
So downloading the `.zip` didn't work, but `git clone`-ing it did work?

Did you `git clone` on Windows? Or on Linux and then copied the files to the Windows machine?

benthebear93, 2021-01-25:
@gavanderhoorn 
So i tried to re-generate the error but couldn't

git clone from window (worked)
download .zip from window(worked)
git clone at linux and move to window machine (worked)
download .zip from linux and move to window machine (worked)

it's weird but everything works for me now.

gavanderhoorn, 2021-01-25:
Hm.

Then unless @SinaMirrazavi tries this as well and reports something else, I don't believe we'll be able to figure out what went wrong.

marshallpowell97, 2021-02-01:
> does SRS / the controller require Windows line-endings?

@gavanderhoorn I don't believe so. If it did, I think we would see more errors than that.

Like you said, the type is provided by the source, so I don't think the problem is in the repo.

gavanderhoorn, 2021-02-19:
Re-opening as it seems others have ran into this now as well.

See #28.

benthebear93, 2021-08-28:
So I encountered same problem today as i tried to change some code in ros_server.
And i took a time to test few things

- using USB to run ros_server NO WORKING
- git clone and download .zip both WORKS
   - using SRS to send the ros_sever&ros_lib to controller and run ros_server WORKING
   - using FTP to send the ros_sever&ros_lib to controller and run ros_server WORKING

i guess using USB dosent work?