ros-industrial/staubli_val3_driver #28: Question - How to transfer val3 from USB to the controller?

Niknu, 2021-02-15:
Hi there,

I'm not sure how I should transfer the val3 driver to the controller, you are writing as followed:

" There are multiple ways of transferring VAL 3 applications to the controller:
1. Copy the contents of val3 folder onto a USB memory stick (<2GB if using CS8), plugging the stick into the controller and using the teach pendant to copy the folders "

In my case, I have a CS8 controller and the controller can see the USB stick with ros_libs + ros_server.
But when I try to load the ros_server it says something with the incorrect type in SimpleRosMsg on the line 1.

I'm doing it correctly, or?

pratipo, 2021-02-16:
hi
exactly the same here, git cloned the repo in ubuntu 18.04, then copied the contents of /val3

controller: cs8c
val: 7.11.2

gavanderhoorn, 2021-02-16:
This starts to sound a bit like #27. Could you both take a look?

Niknu, 2021-02-19:
I have tried to clone and download zip on Linux and Windows. It doesn't work with the #27 solution, 

What I have is :

Controller : cs8
System: s.7.9.1

Below is images of what I assume is needed information

[image]
[image]
[image]

### **_When I see the Event loggers output_**

[image]
[image]
[image]

gavanderhoorn, 2021-02-19:
So both with the `.zip` as with `git clone` gives you these errors?

gavanderhoorn, 2021-02-19:
As this seems to be the same -- or a similar enough -- error as in #27, I'd like to suggest we re-open #27 and continue there.

gavanderhoorn, 2021-02-25:
I'm going to close this one, we'll continue in #27.

closed 2021-02-25 by gavanderhoorn