ros-industrial/staubli_val3_driver #29: error running ros_server

pratipo, 2021-02-16:
hi

i'm trying to get connected to a tx60 using this packet and staubli_experimental one
i'm using melodic

afterwards i'll try to implement a followjointtrajectory action client
is it possible to do it in roslibpy (using rosbridge_server)? i have a blender addon (that i'm going to opensource) that can _stream_  to a topic (so it shouldn't be a problem to send a followJointTrajectoryGoal action) data with joint positions of a simulated/animated tx60

anyway
i uploaded the val3 contents (2 folders) to the cs8c controller using ftp
i set up the sockets as instructed in the readme
[image]

but i guet this error
[image]
it says (i cannot change the controler language :$) "I/O not attached/related to a physical I/O"

stil i'm allowed to run the ros_server program

i got two errors when trying to launch
`roslaunch staubli_val3_driver robot_interface_streaming.launch robot_ip:=192.168.1.10`

so i've pushed the `robot definition` launching load_tx60.launch and the 'controller_joint_names' using
`rosparam set /controller_joint_names "['joint_1', 'joint_2', 'joint_3', 'joint_4', 'joint_5', 'joint_6']"`

then i relaunch the streaming but i guet this

```
SUMMARY

PARAMETERS
 * /robot_ip_address: 192.168.1.10
 * /rosdistro: melodic
 * /rosversion: 1.14.10

NODES
  /
    joint_trajectory_action (industrial_robot_client/joint_trajectory_action)
    motion_streaming_interface (industrial_robot_client/motion_streaming_interface)
    robot_state (industrial_robot_client/robot_state)

ROS_MASTER_URI=http://localhost:11311

process[robot_state-1]: started with pid [5890]
process[motion_streaming_interface-2]: started with pid [5891]
process[joint_trajectory_action-3]: started with pid [5892]
[ WARN] [1613508008.990615796]: Trajectory state not received for 1.000000 seconds
[ERROR] [1613508010.033022744]: Failed to connect to server, rc: -1. Error: 'Connection refused' (errno: 111)
[ERROR] [1613508010.033412197]: Failed to connect to server, rc: -1. Error: 'Connection refused' (errno: 111)
[ERROR] [1613508010.043669608]: Failed to connect to server, rc: -1. Error: 'Connection refused' (errno: 111)
[ WARN] [1613508010.043758701]: Not connected, bytes not sent
[ERROR] [1613508010.043816308]: Failed to receive message length
[ERROR] [1613508010.043852520]: Failed to receive incoming message
[ WARN] [1613508010.043889148]: Send failure, no callback support
[ERROR] [1613508010.066203278]: Failed to connect to server, rc: -1. Error: 'Connection refused' (errno: 111)
[ERROR] [1613508010.316507185]: Timeout connecting to robot controller.  Send new motion command to retry.
[ERROR] [1613508015.046796170]: Failed to connect to server, rc: -1. Error: 'Connection refused' (errno: 111)

```

is there any specific configuration to be set in the pendant related to sockets? i'm trying all of this in manual mode (i need to find out how to trick the door security)

i did the tudelft mooc in a week during christmas and i have no previous experience with ros, so i'm missing lots of basic things (most probably)
thank you very much!!!
Raul

```
tx60 + csc8
ros melodic + ubuntu 18.04.5 lts
```

gavanderhoorn, 2021-02-17:
The error messages shown indicate that either:

 1. the program is not running on the CS8, or
 1. the IP address you've passed to `robot_interface_streaming.launch` is not correct, or
 1. there is a network connection problem between your ROS PC and the CS8, or
 1. there is a firewall between you and the CS8

It's been a long time since I've run this myself, but the error about the IO element could mean the single IO reference we have in the code does not work for your controller/configuration:

https://github.com/ros-industrial/staubli_val3_driver/blob/7913f147663f6d54279cfb77cf90ecfb32454313/staubli_val3_driver/val3/ros_server/ros_server.dtx#L34-L36

Perhaps @marshallpowell97 recognises the error?

pratipo, 2021-02-17:
i'll check first the io, altough i think the program runs anyway

> It's been a long time since I've run this myself, but the error about the IO element could mean the single IO reference we have in the code does not work for your controller/configuration:

do you think i can just remove those 3 lines from `ros_server.dtx` ?

the controller i have is a CS8**C**, i'm not sure it makes any difference

thanks again

gavanderhoorn, 2021-02-17:
>  altough i think the program runs anyway

the error messages indicate the required TCP ports aren't actually open/accepting connections.

The most seen cause for that is the server program not actually running.

> do you think i can just remove those 3 lines from `ros_server.dtx` ?

I'm not sure sure. It was added by @marshallpowell97 in #8 (in 4cbb7f92), but that may have been an auto-update. Those lines have been part of the driver since the initial import though:

https://github.com/ros-industrial/staubli_val3_driver/blob/3978dd3ed794dba69c03290471800141e8c19312/staubli_val3_driver/val3/ros_server/ros_server.dtx#L34-L36

ilaro-org, 2021-02-18:
hi

I sorted out, when configuring the sockets ("Feedback" and "Motion") i didnt know how to use caps
i modified ros_server.dtx and it worked, so i reversed the changes and used this

[image]

the i/o error is gone 

and i can get the joint positions by echoing /joint_states topic :)

thanks so much!

gavanderhoorn, 2021-02-18:
Ok, thanks for reporting back.

I'll close this then.

gavanderhoorn, 2021-02-18:
(I'm guessing you used the wrong account to post your comment, and you're actually @pratipo)

closed 2021-02-18 by gavanderhoorn

ilaro-org, 2021-02-18:
oh sorry, yes its pratipo

i actullay got some extra errors after reuploading the fresh val3 folders (just in case)
i fixed it by 
* git cloning (instead of downloading a zip)
* chmod 777 all the files in /val3

i'm not sure whether both or just one of the above makes it actually work, i could actually try to find out, if you think it's usefull

thanks again

gavanderhoorn, 2021-02-18:
> i fixed it by
> 
> -  git cloning (instead of downloading a zip)

hm. That reminds me of #27.

It would be great if you could test a `.zip` and see whether that makes any difference.

I would not expect `chmod 777` to matter: there aren't that many FTP clients which preserve access permissions on uploaded files.