ros-industrial/staubli_val3_driver #33: ros_server NOT running in CS9 controller

AshwathamanGovind, 2021-06-09:
When i tried to start the ros_server after sending the contents inside the val3 directory to the usrapp/ directory in cs9. An error saying "cannot start ros_server.pgx" is shown and i am not able to start the server. I am using a cs9 controller and TX2_60 Robot here and i am making use of the staubli driver that i got from https://github.com/ros-industrial/staubli_val3_driver/ . Can you anyone help me sort out this problem and establish a seamless communication between ROS and staubli cs9 ??? 

Thanks in Advance. Any help from your side would be of great use for me

gavanderhoorn, 2021-06-09:
As far as we know, CS9 should be supported and work. This was contributed by @marshallpowell97 in #15.

Are there are any more informative (error) messages displayed then `cannot start ros_server.pgx`?

Perhaps you could take a picture of the TP when it shows you that error?

AshwathamanGovind, 2021-06-10:
This is error that i get when i try to run the ros_server from SP2 in Staubli 
[image]

JohannesFriedrich, 2021-06-10:
Just a little hint from my side: 
You also have to load the "ros_libs" folder to usrapp. It looks like in your image that this library is not on the disk.

AshwathamanGovind, 2021-06-10:
@JohannesFriedrich 
I have loaded that ros_libs folder also to usrapp. I also checked that again using ftp. The folder is present inside usrapp but it is not listed in the VAL3 APPLICATIONS tab in the teach pendant. I have attached the snapshot of the file structure (accessed using FTP).
[image]

JohannesFriedrich, 2021-06-10:
What´s the error logger (upper left symbol) exactly saying? Can you give us a screenshot or better: an exported error-logger? You can export it also via ftp by changing the directory to _log_ or export it via USB by exporting it in the error-logger panel, click right on the little arrow (sub-menu) and export it.

AshwathamanGovind, 2021-06-10:
Please have a look. I have attached the exported error log folder that i have exported from CS9. 
[exportLogs.zip](https://github.com/ros-industrial/staubli_val3_driver/files/6631692/exportLogs.zip)

AshwathamanGovind, 2021-06-10:
While going through the error logs I encountered another error message saying "interfaceDefault.pjx: invalid library". FYI, i will also add the screenshot of that error message. I think due to some error in the libs I am able to start the ROS server in CS9.
[image]

JohannesFriedrich, 2021-06-10:
Indeed, the _UserInterface_ library is the problem. Did you change something in this library? I think the best way is to remove the _UserInterface_ completely from the controller and copy it again from your PC.

AshwathamanGovind, 2021-06-11:
As expected, The issue was because of a corrupt file in UserInterface library. After i sent that via again via Transfer manager now I am able to start the ros_server application. But i am facing an issue while i try to connect to the robot from ROS. I getting this error:- 

 **Exception while loading database plugin 'warehouse_ros_mongo::MongoDatabaseConnection': According to the loaded plugin descriptions the class warehouse_ros_mongo::MongoDatabaseConnection with base class type warehouse_ros::DatabaseConnection does not exist. Declared types are** 

Any idea about this??

AshwathamanGovind, 2021-06-11:
Thanks for the support @gavanderhoorn and @JohannesFriedrich

closed 2021-06-11 by AshwathamanGovind