ros-industrial/staubli_val3_driver #34: Problems in understanding and troubleshooting Staubli TX2_60 with CS9

AshwathamanGovind, 2021-06-10:
Hai,
I am new to ROS development and Staubli robots. Though i have some prior knowledge on UR5e, I am finding it hard to understand the architecture of Staubli and its other programming interfaces (like staubli robotics suite). So can anyone suggest a beginners tutorial or a guide where i can get a good understanding on the staubli's architecture(VAL3, SRS) and its ROS interfacing. I am not able to find much information or troubleshooting tutorials in the internet (no idea why). It looked like a very closed loop for a beginner like me. Any help from your side would be very helpful. 

Thanks in advance. I think this question is a bit out of context with respect to this forums. As i can't find any good forums (like UR forums) for Staubli (please suggest one,if anyone knows) , I thought this would be helpful for me.

gavanderhoorn, 2021-06-10:
You're correct: your question is off-topic for this issue tracker.

I can only recommend to reach out to your local Staubli branch office to ask them about training.

As with all `ros-industrial` robot drivers: `staubli_val3_driver` should *not* be your first exposure/interaction with an industrial robot.

Too many things can go wrong, and you'll be trying to learn many different things at the same time, which may not be the most efficient way to go about it.

We do not have any beginners tutorials, as we expect the user to have the necessary experience to not need those.

That sounds harsh, I know, but resources are limited, and we are not a Staubli training centre.

gavanderhoorn, 2021-06-10:
As this is not an issue with the package in this repository, I'm going to close the issue.

closed 2021-06-10 by gavanderhoorn