ros-industrial/staubli_val3_driver #37: How to make it work?

lepalom, 2021-06-30:
One year ago I install this driver, and it worked. Now I'm try to repeat the same, but with Noetic and I have some troubles. I'm using Debian Buster. Steps that i have done:

- clone the repo and create a catkin workspace and also clone industrial_core.
- build.
- install the val3 driver to the controller following the readme instructions.
- run in the controller the val3_server
- launch in one terminal:
  $ roslaunch staubli_tx90_support load_tx90.launch

So, robot description is loaded.

- In another terminal, 
 $ roslaunch staubli_val3_driver motion_streaming_interface.launch robot_ip:=x.x.x.x

In the controller appears the Motion is connected, but Feedback no. If I run:
$ rostopic echo /joint_states

Nothing appears. Do I have missed some step? Is this a correct behaviour?
I remember that I could see the robot in rviz and when I moved with the teach pendant rviz will update it.

gavanderhoorn, 2021-06-30:
The correct `.launch` file would be [robot_interface_streaming.launch](https://github.com/ros-industrial/staubli_val3_driver/blob/7913f147663f6d54279cfb77cf90ecfb32454313/staubli_val3_driver/launch/robot_interface_streaming.launch).

`motion_streaming_interface.launch` only starts the trajectory relay.

gavanderhoorn, 2021-06-30:
I'm closing this as I assume we've resolved the problem.

Feel free to keep commenting on the issue.

closed 2021-06-30 by gavanderhoorn

lepalom, 2021-07-01:
Thanks @gavanderhoorn . It was my mistake. Sorry.

gavanderhoorn, 2021-07-01:
No problem.