ros-industrial/staubli_val3_driver #40: Running the driver without teach pendant

lepalom, 2021-07-21:
We are trying to run the val3_driver without the MCP. We have *only* one to control two robots. We have detected that, we can run it remotely without any problem, but with the robot with teach pendant attached. 

Looking on the logs, we have found a  "Internal error MCP-NoAnswer: char 1" and checking the val3 code, in the start:

`taskCreate sScreenTaskName, 10, screenUpdate()
`

so, it's the task to create the update on the screen. So, the question is:

* Can we run the val3 driver without teach pendant?
* It is possible that that task affect and cannot load the ros_server application?
* It's safe to remove that task?

Thanks in advance.

JohannesFriedrich, 2021-07-22:
Hello @lepalom,

I think it´s clear that you have to use a dummy-plug if you remove the MCP. Do you use it and mounted it correct?
The task should not make any problems either with or without MCP. 
Are there any runtime errors in the task? You can find these error in the logger or with the online-debugger of SRS.

lepalom, 2021-07-22:
@JohannesFriedrich , you are right. I lost one morning checking all and not found any solution. This morning, I disassembled the dummy-plug and found that it was wrong. So, after mount again, we could make the test, and it's working.

Thanks for the answer anyway.

closed 2021-07-22 by lepalom