ros-industrial/staubli_val3_driver #42: val3 controller only get limited trajectory into buffer. (unable to fully excute the trjaectory)

benthebear93, 2021-08-21:
Hello
(env : 18.04, melodic, Tx90)

i been using this pkg to send trajectory to TX90.
Every thing seems working fine till i tried to send long trajectory.
robot_interface_streaming shows that all the Points are sent to controller, but my pendant only got few of the points

here is roslaunch --screen log.

`[ INFO] [1629529937.060229013]: Using joint names from URDF: 'robot_description': [joint_1, joint_2, joint_3, joint_4, joint_5, joint_6]
[ INFO] [1629529937.060520612]: Filtered joint names to 6 joints
[ INFO] [1629529937.082788623]: Unlocking mutex
[ INFO] [1629529937.082877382]: Starting joint trajectory streamer thread
[ INFO] [1629529937.088098549]: Connecting to robot motion server
[ WARN] [1629529937.088168345]: Tried to connect when socket already in connected state
[ WARN] [1629529938.083812518]: Trajectory state not received for 1.000000 seconds
[ INFO] [1629529959.361065248]: Received new goal
[ INFO] [1629529959.361299378]: Publishing trajectory
[ INFO] [1629529959.362446511]: Receiving joint trajectory message
[ WARN] [1629529959.363097194]: Joint velocity-limits unspecified.  Using default velocity-ratio.
[ INFO] [1629529959.366991150]: Loading trajectory, setting state to streaming
[ INFO] [1629529959.367063829]: Executing trajectory of size: 122
[ INFO] [1629529959.380736553]: Point[1 of 122] sent to controller
[ INFO] [1629529959.380896108]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629529959.395165687]: Point[2 of 122] sent to controller
[ INFO] [1629529959.399821465]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629529959.411128088]: Point[3 of 122] sent to controller
[ INFO] [1629529959.419703071]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629529959.427121521]: Point[4 of 122] sent to controller
[ INFO] [1629529959.443115726]: Point[5 of 122] sent to controller
[ INFO] [1629529959.445508169]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629529959.460490154]: Point[6 of 122] sent to controller
[ INFO] [1629529959.460756840]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629529959.475086737]: Point[7 of 122] sent to controller
[ INFO] [1629529959.479579891]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629529959.491057778]: Point[8 of 122] sent to controller
[ INFO] [1629529959.499626569]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629529959.507063767]: Point[9 of 122] sent to controller
[ INFO] [1629529959.523044818]: Point[10 of 122] sent to controller
[ INFO] [1629529959.525322925]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629529959.540103078]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629529959.541540422]: Point[11 of 122] sent to controller
[ INFO] [1629529959.560087773]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629529959.560406430]: Point[12 of 122] sent to controller
[ INFO] [1629529959.575107509]: Point[13 of 122] sent to controller
[ INFO] [1629529959.579596198]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629529959.591062971]: Point[14 of 122] sent to controller
[ INFO] [1629529959.599562494]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629529959.607095441]: Point[15 of 122] sent to controller
[ INFO] [1629529959.623256708]: Point[16 of 122] sent to controller
[ INFO] [1629529959.625323065]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629529959.640665806]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629529959.641507269]: Point[17 of 122] sent to controller
[ INFO] [1629529959.660506384]: Point[18 of 122] sent to controller
[ INFO] [1629529959.660890278]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629529959.674982773]: Point[19 of 122] sent to controller
[ INFO] [1629529959.679675191]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629529959.691013872]: Point[20 of 122] sent to controller
[ INFO] [1629529959.699563661]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629529959.707057697]: Point[21 of 122] sent to controller
[ INFO] [1629529959.719614525]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629529959.724507525]: Point[22 of 122] sent to controller
[ INFO] [1629529959.740673843]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629529959.741521878]: Point[23 of 122] sent to controller
[ INFO] [1629529959.754961963]: Point[24 of 122] sent to controller
[ INFO] [1629529959.759517107]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629529959.770913574]: Point[25 of 122] sent to controller
[ INFO] [1629529959.779459494]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629529959.787030686]: Point[26 of 122] sent to controller
[ INFO] [1629529959.803007862]: Point[27 of 122] sent to controller
[... 164 lines cut ...]
[ INFO] [1629529961.264414936]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629529961.279560052]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629529961.279771324]: Point[119 of 122] sent to controller
[ INFO] [1629529961.294353004]: Point[120 of 122] sent to controller
[ INFO] [1629529961.298719989]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629529961.310354603]: Point[121 of 122] sent to controller
[ INFO] [1629529961.318956515]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629529961.326301237]: Point[122 of 122] sent to controller
[ INFO] [1629529961.331533733]: Trajectory streaming complete, setting state to IDLE
[ INFO] [1629529961.338989305]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629529961.358951248]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629529961.378899899]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629529961.398910445]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629529961.418926000]: Waiting to check for goal completion until halfway through trajectory
`

122 points are sent to controller but my pendant only shows 31.
Trajectory buffer : 10  21
Motion buffer      : 21   0

This seems similar to #25.
but little different since my pendant only shows maximum 31.
could it be the val3 driver problem ? or is there any setting in the pendant that i don't know about?

Thank you for your help, if more information needed let me know!

gavanderhoorn, 2021-08-21:
It could be it's a bug in the VAL 3 code, we don't know right now.

As in https://github.com/ros-industrial/staubli_val3_driver/issues/25#issuecomment-737170702, please provide:

> Please provide the following:
> 
>  1. a full copy-paste of the output on your console after `roslaunch`ing your system (append `--screen` to the `roslaunch` command so all the output is shown)
> 
>  2. ideally some code which shows how you generate your trajectories, and how you submit them to the driver (**note**: we don't need a full application. Just the segment(s) which is (are) responsible for putting together the actual trajectory and the goal)
> 
>  3. the ROS `JointTrajectory` which gets submitted as part of the action goal to the driver
> 
>  4. a capture of the network traffic between the ROS pc and the CS9
> 
> 
> The trajectory you should be able to capture using `rostopic echo` on the topic which carries the goal message.
> 
> Network traffic can be captured using Wireshark.

benthebear93, 2021-08-21:
I will try to provide all information asap

**1. Here is full copy-pase of output console of roslaunch** 
```
roslaunch staubli_val3_driver robot_interface_streaming.launch robot_ip:=192.168.0.254 --screen
... logging to /home/benlee/.ros/log/8e399e10-024e-11ec-aab8-bc542fd927d1/roslaunch-benlee-24844.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://localhost:43057/

SUMMARY
========

PARAMETERS
 * /robot_ip_address: 192.168.0.254
 * /rosdistro: melodic
 * /rosversion: 1.14.11

NODES
  /
    joint_trajectory_action (industrial_robot_client/joint_trajectory_action)
    motion_streaming_interface (industrial_robot_client/motion_streaming_interface)
    robot_state (industrial_robot_client/robot_state)

ROS_MASTER_URI=http://localhost:11311

process[robot_state-1]: started with pid [24868]
process[motion_streaming_interface-2]: started with pid [24869]
process[joint_trajectory_action-3]: started with pid [24870]
[ INFO] [1629530593.137128749]: Added message handler for message type: 0
[ INFO] [1629530593.146347433]: Robot state connecting to IP address: '192.168.0.254:11002'
[ WARN] [1629530593.147505367]: Unable to find user-specified joint names in 'controller_joint_names'
[ INFO] [1629530593.156119580]: Joint Trajectory Interface connecting to IP address: '192.168.0.254:11000'
[ INFO] [1629530593.158700436]: Using joint names from URDF: 'robot_description': [joint_1, joint_2, joint_3, joint_4, joint_5, joint_6]
[ WARN] [1629530593.159117710]: Unable to find user-specified joint names in 'controller_joint_names'
[ INFO] [1629530593.159510768]: Connected to server
[ INFO] [1629530593.159574885]: Initializing message manager with default comms fault handler
[ INFO] [1629530593.159634989]: Default communications fault handler successfully initialized
[ INFO] [1629530593.159677329]: Initializing message manager
[ INFO] [1629530593.159720840]: Added message handler for message type: 1
[ INFO] [1629530593.164207204]: Added message handler for message type: 10
[ INFO] [1629530593.165791538]: Entering message manager spin loop
[ INFO] [1629530593.166808874]: Using joint names from URDF: 'robot_description': [joint_1, joint_2, joint_3, joint_4, joint_5, joint_6]
[ INFO] [1629530593.167100677]: JointTrajectoryStreamer: init
[ INFO] [1629530593.167712367]: Connected to server
[ WARN] [1629530593.180047068]: Unable to find user-specified joint names in 'controller_joint_names'
[ INFO] [1629530593.192670085]: Using joint names from URDF: 'robot_description': [joint_1, joint_2, joint_3, joint_4, joint_5, joint_6]
[ INFO] [1629530593.192923402]: Filtered joint names to 6 joints
[ INFO] [1629530593.195012085]: Unlocking mutex
[ INFO] [1629530593.195118255]: Starting joint trajectory streamer thread
[ INFO] [1629530593.200348580]: Connecting to robot motion server
[ WARN] [1629530593.200496628]: Tried to connect when socket already in connected state
[ INFO] [1629530606.440990664]: Received new goal
[ INFO] [1629530606.441153604]: Publishing trajectory
[ INFO] [1629530606.441969782]: Receiving joint trajectory message
[ WARN] [1629530606.442407752]: Joint velocity-limits unspecified.  Using default velocity-ratio.
[ INFO] [1629530606.444867723]: Loading trajectory, setting state to streaming
[ INFO] [1629530606.444951228]: Executing trajectory of size: 105
[ INFO] [1629530606.456190220]: Point[1 of 105] sent to controller
[ INFO] [1629530606.460496045]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629530606.472175343]: Point[2 of 105] sent to controller
[ INFO] [1629530606.480503701]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629530606.488074592]: Point[3 of 105] sent to controller
[ INFO] [1629530606.504004485]: Point[4 of 105] sent to controller
[ INFO] [1629530606.506258921]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629530606.521514402]: Point[5 of 105] sent to controller
[ INFO] [1629530606.521876024]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629530606.536025871]: Point[6 of 105] sent to controller
[... 606 lines cut ...]
    effort: []
    time_from_start: 
      secs: 4
      nsecs: 742362012
  - 
    positions: [-1.536881896597685e-05, 2.2318234855807875, 0.32381667666380304, 6.96343712493885e-06, -0.984759212756878, -4.053512104618042e-05]
    velocities: [9.55273098499381e-07, 0.15615952752451545, -0.38248982292525924, 2.0652223531889623e-06, 0.22633029539238847, -2.3687847265490806e-06]
    accelerations: [-3.293337571034941e-06, -0.3713084156837698, 0.9792913832428255, -5.836725652044982e-06, -0.607982967537173, 6.156734039297687e-06]
    effort: []
    time_from_start: 
      secs: 4
      nsecs: 891881394
  - 
    positions: [-1.51593574306374e-05, 2.278405146432451, 0.21487989102713972, 7.51106784345107e-06, -0.9224040879742216, -4.1202983641294254e-05]
    velocities: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
    accelerations: [-1.86501071585009e-06, -0.41475537028628007, 0.9699550432177863, -4.876012947669871e-06, -0.5551996729106404, 5.9465376045496306e-06]
    effort: []
    time_from_start: 
      secs: 5
      nsecs: 365824787
```

I will try to fill more information about network traffic, since i never used Wireshark, with roslaunch log and JointTrajectory echo data again, (probably tomorrow).

Thx for the help!

benthebear93, 2021-08-22:
**1. Here is full copy-pase of output console of roslaunch**
```
r goal completion until halfway through trajectory
[ INFO] [1629608810.666186548]: Point[13 of 114] sent to controller
[ INFO] [1629608810.682331258]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629608810.683691826]: Point[14 of 114] sent to controller
[ INFO] [1629608810.702085100]: Point[15 of 114] sent to controller
[ INFO] [1629608810.702391555]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629608810.716748357]: Point[16 of 114] sent to controller
[ INFO] [1629608810.721112264]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629608810.732569690]: Point[17 of 114] sent to controller
[ INFO] [1629608810.741179052]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629608810.748763719]: Point[18 of 114] sent to controller
[ INFO] [1629608810.764563440]: Point[19 of 114] sent to controller
[ INFO] [1629608810.766924772]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629608810.782313189]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629608810.782888979]: Point[20 of 114] sent to controller
[ INFO] [1629608810.801894944]: Point[21 of 114] sent to controller
[ INFO] [1629608810.802139261]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629608810.816646886]: Point[22 of 114] sent to controller
[ INFO] [1629608810.820894257]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629608810.832511140]: Point[23 of 114] sent to controller
[ INFO] [1629608810.840955750]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629608810.848429705]: Point[24 of 114] sent to controller
[ INFO] [1629608810.864509467]: Point[25 of 114] sent to controller
[ INFO] [1629608810.866335132]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629608810.882084967]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629608810.882927572]: Point[26 of 114] sent to controller
[ INFO] [1629608810.902025955]: Point[27 of 114] sent to controller
[ INFO] [1629608810.902124442]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629608810.916453629]: Point[28 of 114] sent to controller
[ INFO] [1629608810.920731308]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629608810.932504217]: Point[29 of 114] sent to controller
[ INFO] [1629608810.940931971]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629608810.948568962]: Point[30 of 114] sent to controller
[ INFO] [1629608810.964497648]: Point[31 of 114] sent to controller
[ INFO] [1629608810.966309006]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629608810.982147629]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629608810.982843206]: Point[32 of 114] sent to controller
[ INFO] [1629608811.001852011]: Point[33 of 114] sent to controller
[ INFO] [1629608811.002034647]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629608811.016441606]: Point[34 of 114] sent to controller
[ INFO] [1629608811.020621362]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629608811.032409516]: Point[35 of 114] sent to controller
[ INFO] [1629608811.040976248]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629608811.048510440]: Point[36 of 114] sent to controller
[ INFO] [1629608811.060627218]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629608811.065978967]: Point[37 of 114] sent to controller
[ INFO] [1629608811.082010393]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629608811.082800905]: Point[38 of 114] sent to controller
[ INFO] [1629608811.101782007]: Point[39 of 114] sent to controller
[ INFO] [1629608811.102021406]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629608811.116344373]: Point[40 of 114] sent to controller
[ INFO] [1629608811.120658092]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629608811.132420063]: Point[41 of 114] sent to controller
[ INFO] [1629608811.140858060]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629608811.148407183]: Point[42 of 114] sent to controller
[ INFO] [1629608811.164315165]: Point[43 of 114] sent to controller
[ INFO] [1629608811.166420610]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629608811.181980416]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629608811.183554999]: Point[44 of 114] sent to controller
[ INFO] [1629608811.201751739]: Point[45 of 114] sent to controller
[ INFO] [1629608811.202022872]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629608811.216476026]: Point[46 of 114] sent to controller
[ INFO] [1629608811.220838236]: Waiting to check for goal completion until halfway through trajectory
[ INFO] [1629608811.232451464]: Point[47 of 114] sent to controller
[ INFO] [1629608811.240774690]: Waiting to check for goal completion until halfway through trajectory
[... cut ...]
    positions: [5.039871312302024e-05, 1.78154344816012, 0.2424484940409218, -8.975688463229239e-05, -0.45322938980605815, 0.0001448845225596476]
    velocities: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
    accelerations: [4.148588366343691e-06, -0.46903371659963855, 0.9914467153540105, 9.146902983986422e-05, -0.5224129966089346, -0.00010142986506173747]
    effort: []
    time_from_start: 
      secs: 6
      nsecs: 264287337
---
```
**4. Wireshark log**
[wireshark_log.zip](https://github.com/ros-industrial/staubli_val3_driver/files/7026623/wireshark_log.zip)

I got the Wireshark network traffic log.

gavanderhoorn, 2021-08-22:
According to the SimpleMessage traffic (in your wireshark capture), the robot never enabled the drives, nor was it taken out of `MANUAL` mode. That would make motion impossible.

Two things:

 1. if you run your robot in `MANUAL` mode, you cannot let go of the teach pendant. Please refer to the Staubli documentation on how to run programs in manual mode
 2. test out ros-industrial/industrial_core#271: it should print an error and refuse to accept the `FollowJointTrajectory` goal when the robot does not report it's capable of executing motion

gavanderhoorn, 2021-08-22:
As I don't believe there is a problem right now with `staubli_val3_driver`, I'm closing this issue.

If it turns out there is a problem, we can re-open.

closed 2021-08-22 by gavanderhoorn

benthebear93, 2021-08-22:
@gavanderhoorn  
Sorry for bad issue. Yes it was `MANUAL` mode problem. changing mode solved problem 
thx for the help!

gavanderhoorn, 2021-08-22:
No problem.

Note: running your program(s) in manual mode is not a problem necessarily.

The problem is not holding the teach pendant (depressing the deadman switch, etc).

That will make the robot controller disable the drives.