ros-industrial/staubli_val3_driver #46: Val3 Driver Version Requirements Qestion

JGittles, 2021-10-01:
Hi All,

I am trying to do realtime control with the FAU fork of ros_Industrial. while running in SRS after acquiring the motion addon from staubli I could not run the sim and noticed that the ros_industrial fork uses val3 s7.9.1 and my controller is running either 7.6 and 7.11.2. do these versions need to be an exact match for successful execution?

JGittles, 2021-10-01:
additional note: I am unsure of the specificity of the motion addon for Val3 version compatibility but I have the one at least intended for s7.6.

gavanderhoorn, 2021-10-05:
I can't comment on the exact version requirement. The work was done in a fork, isolated.

The original author(s) would have to provide that information.

gavanderhoorn, 2021-10-05:
Perhaps @enginkarlidag can comment.

JGittles, 2021-10-05:
[image]

here is also a relevant image of the error

enginkarlidag, 2021-10-05:
The VAL3 version could be anything from s7.7.2 or greater (for CS8 controllers). The minimum version is given as a requirement as described in the [README](https://github.com/ros-industrial/staubli_val3_driver/blob/master/staubli_val3_driver/README.md#requirements) of this repo.

And just to avoid misunderstandings: The following information mainly refers to our fork of the driver!

So our project seems to be saved with version s7.9.1 but it should be possible to use any version (>= s7.7.2) to run it. You just have to make sure that the required add-on is installed and the sockets are configured as described in the [README](https://github.com/FAU-FAPS/staubli_val3_driver/tree/faps-devel/staubli_val3_driver#configuration).

Afaik, you always need the corresponding add-on(s) for a specific VAL3 version. For example, if you want to use s7.11.2 then you should get and install the add-on for this version. Btw, you must also enable the add-on for your controller in SRS explicitly. You can do that directly in the "New Cell Wizard" while creating a new cell or afterwards in the controller configuration. Just right-click on the controller in the "Cell Explorer" and go to "Controller Configuration". There you should have a tab called "Add Ons" if you have installed the correct version of the add-on. I was able to reproduce your error when the add-on was disabled. So enabling the add-on should fix it. This is only required for the emulated controller in SRS. For the real controller I think you get .so files which you have to copy via an FTP connection to the controller and they should be "enabled" after a reboot automatically.

[image]

closed 2021-10-12 by JGittles

JGittles, 2021-10-13:
@gavanderhoorn yes Engin has fixed my issue and I was successfully able to run this in srs. issue has been closed