ros-industrial/staubli_val3_driver #48: Scara support

simonschmeisser, 2021-12-09:
Has anyone experimented with using/extending this driver for 4-axis scara robots? We're looking at the TS2-100 to be specific but that shouldn't matter in general

gavanderhoorn, 2021-12-09:
See https://github.com/ros-industrial/staubli_experimental/issues/36.

simonschmeisser, 2021-12-09:
Thanks for the link! Good to know that my optimistic guess of this being achievable was not too optimistic. If you still have the code around somewhere that would be great but the overview in that comment is also already a good start. Let's see if the project materializes first.

gavanderhoorn, 2021-12-09:
I'll see if I can find it.

It should be somewhere on a laptop I used, but *where* ..

gavanderhoorn, 2021-12-09:
I do remember the scara having more trouble with the acceleration issues we still have with `staubli_val3_driver` though.

gavanderhoorn, 2021-12-09:
I believe https://github.com/ros-industrial/staubli_val3_driver/compare/master...gavanderhoorn:wip_scara_support should do it.

No guarantees of course.

And heed the 'warning':

> ugly, hacky ..

gavanderhoorn, 2022-02-09:
Have you been able to verify this works @simonschmeisser ?

simonschmeisser, 2022-02-09:
No, not yet unfortunately. The project is still in the planning phase currently and I have not yet seen any hardware unfortunately.

gavanderhoorn, 2022-05-02:
Friendly ping